arkhipenko/esp32-cam-mjpeg-multiclient #8: features

mplaxman, 2021-01-07:
hi i want use esp eye as realtime face recognization can i use this ?

if yes how to train bulk images in this ? is it possible to train the model in esp eye or we need train the model in server ?

we need some help if u give it is very helpful for me

arkhipenko, 2021-01-07:
Face recognition was not part of this project, sorry.
That would be a separate project.

closed 2021-01-07 by arkhipenko

mplaxman, 2021-01-07:
we know we are  asking suggestion if u have any information please share

arkhipenko, 2021-01-08:
Unfortunately not. This could be a good paid research project though if you are interested.

mplaxman, 2021-01-09:
what are the charges for this project ?

arkhipenko, 2021-01-09:
Let's move this conversation to the email please. Send me a direct email to arkhipenko@hotmail.com