arkhipenko/esp32-cam-mjpeg-multiclient #10: Problem getting static jpg image while steaming mjpg

goebish, 2021-06-03:
Hi, there's an issue capturing jpg image while another client is streaming mjpg, most of the time the jpg image is garbled:
https://imgur.com/a/KVj6g8p
and it often stops mjpg streaming on the other client altogether with an error in the log (have to reset the esp32 to make it work again)
[E][camera.c:1483] esp_camera_fb_get(): Failed to get the frame on time!

arkhipenko, 2021-06-03:
I actually never tested this scenario. Will have a look when I have some time.

arkhipenko, 2021-07-01:
The problem is that a static JPEG stream

https://github.com/arkhipenko/esp32-cam-mjpeg-multiclient/blob/master/esp32_camera_mjpeg_multiclient.ino#L357

was never moved to a separate thread from the server, since this was not the purpose of the POC. 

If and when I have the time (unlikely) I might move it to something like this:

https://github.com/arkhipenko/esp32-cam-mjpeg-multiclient/blob/master/esp32_camera_mjpeg_multiclient.ino#L282

OR you can try to do it!

closed 2021-07-01 by arkhipenko

goebish, 2021-07-01:
Ok, I'll try to make sense of this!

arkhipenko, 2021-07-01:
If you need this functionality for a commercial project you can always hire me for a custom solution.

goebish, 2021-07-01:
No, that's just for using with my 3D printer ;)