arkhipenko/esp32-cam-mjpeg-multiclient #12: How to rotate the screen ?

yukiloh, 2022-06-08:
Like`document.body.style.transform = "rotate(90deg)";`
How to change the code?
Thanks

arkhipenko, 2022-06-08:
There are no 90 or 270 degree transforms. Only vertical/horizons flips are available to the best of my knowledge (sample code located and commented out right after the camera initialization)

closed 2022-06-10 by arkhipenko

serbasan, 2024-03-16:
Hello. I can't find where to rotate the image 180 degrees. Could you help me? Thank you

serbasan, 2024-03-17:
Hello. I answer myself. It would be here:

  if (cam.init(config) != ESP_OK) {
     Serial.println("Error initializing the camera");
     delay(10000);
     ESP.restart();
   }
   sensor_t * s = esp_camera_sensor_get();
   s->set_vflip(s, 1); // 0 = disable , 1 = enable
  
   // Configure and connect to WiFi
   IPAddress ip;